Seeed-Studio/seeed-linux-dtoverlays #103: make all_rpi fails

cipy, 2024-05-26:
**Describe the bug**
up to date kernel and RPi on bullseye

**To Reproduce**
Steps to reproduce the behavior:
1. read readme.md

**Expected behavior**
it works

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Debian Bullseye
 - Version: updated today

**Additional context**

> pi@raspberrypi:~/work/seeed-linux-dtoverlays $ make all_rpi
>   DTC     overlays/rpi/reComputer-R100x-overlay.dtbo
>   DTC     overlays/rpi/lis331dlh-overlay.dtbo
>   DTC     overlays/rpi/reTerminal-plus-overlay.dtbo
>   DTC     overlays/rpi/2xMCP2517FD-overlay.dtbo
> overlays/rpi/2xMCP2517FD-overlay.dts:93.25-102.15: Warning (spi_bus_reg): /fragment@5/__overlay__/can@1: SPI bus unit address format error, expected "0"
>   DTC     overlays/rpi/grove-led-overlay.dtbo
> overlays/rpi/grove-led-overlay.dts:33.3-32: Warning (label_is_string): /__overrides__:label: property is not a string
> overlays/rpi/grove-led-overlay.dts:30.3-31.26: Warning (gpios_property): /__overrides__:gpio: property size (22) is invalid, expected multiple of 4
>   DTC     overlays/rpi/tpm-spi-slx9670-overlay.dtbo
>   DTC     overlays/rpi/bma456-i2c1-overlay.dtbo
>   DTC     overlays/rpi/2xMCP2518FD-spi0-overlay.dtbo
> overlays/rpi/2xMCP2518FD-spi0-overlay.dts:59.25-68.15: Warning (spi_bus_reg): /fragment@4/__overlay__/can@0: SPI bus unit address format error, expected "1"
> overlays/rpi/2xMCP2518FD-spi0-overlay.dts:70.25-78.15: Warning (spi_bus_reg): /fragment@4/__overlay__/can@1: SPI bus unit address format error, expected "0"
> make[1]: *** /lib/modules/6.1.21-v8+/build: No such file or directory.  Stop.
> make: *** [Makefile:157: all_rpi] Error 2

cipy, 2024-05-26:
`pi@raspberrypi:~/work/seeed-linux-dtoverlays/scripts $ sudo ./reTerminal.sh --compat-kernel` ends in

> KBUILD: /lib/modules/5.10.17-v7l+/build
> cp: cannot stat '/home/pi/work/seeed-linux-dtoverlays/scripts/modules/mipi_dsi/*': No such file or directory
> Error! Could not locate dkms.conf file.
> File: /usr/src/mipi_dsi-0.1/dkms.conf does not exist.
> Can't compile with this kernel, aborting
> Please try to compile with the option --compat-kernel
> cat: /var/lib/dkms/mipi_dsi/0.1/build/make.log: No such file or directory

bigbearishappy, 2024-05-27:
Actually, you should run `sudo ./scripts/reTerminal.sh --compat-kernel`.
Can you have a try and attatch the full log of it in here?

bigbearishappy, 2024-05-27:
`sudo ./scripts/reTerminal.sh --keep-kernel` should also work fine too

bigbearishappy, 2024-07-02:
- Closing the issue. If there are any further problem. Please reopen it.

closed 2024-07-02 by bigbearishappy