Seeed-Studio/seeed-linux-dtoverlays #106: Jetson nano 4gb dev kit , MCP251x didn’t enter in conf mode after reset in jetpack 4.6.1

stepin104954, 2024-08-04:
Subject: Issue with Implementing CAN Connection on Jetson Nano 4GB with (MCP251x) using CAN SPI Click 3.3V shield.

Hello,

I am facing issues while implementing a CAN connection on my Jetson Nano 4GB Developer Kit using the CAN SPI Click 3.3V shield - [CAN SPI click 3.3v - Breakout board for SN65HVD230 CAN transciever](https://www.mikroe.com/can-spi-33v-click) .

**System Details**:

* L4T Version: 32.7.1
* JetPack Version: 4.6.1

**Steps Followed:**
Followed the steps from this post, just edited the oscillator value to 10 MHz to match the spi click shield from Mikro E.

[quote]
Please follow:
[/quote]

**Steps :**

1. Downloaded JetPack 4.6.1 image and flashed it using Balena Etcher.
2. After setting up the system and connecting to Wi-Fi, I followed these steps:

**Cloning Repository:**

```
git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays
cd seeed-linux-dtoverlays
```

**Creating DTS File:**

* Created a `jetson-mcp251x.dts` file with the following code and saved it in the `seeed-linux-dtoverlays/overlays/jetsonnano` directory:

```
/dts-v1/;
/plugin/;

#include <dt-bindings/pinctrl/pinctrl-tegra.h>
#include <dt-bindings/gpio/tegra-gpio.h>

/ {
    overlay-name = "MCP251x CAN Controller";
    compatible = "nvidia,p3449-0000-b00+p3448-0000-b00";

    fragment@0 {
        target-path = "/";
        __overlay__ {
            clocks {
                can_clock: can_clock {
                    compatible = "fixed-clock";
                    #clock-cells = <0>;
                    clock-frequency = <10000000>;      
                    clock-accuracy = <100>;
                };
            };
        };
    };

    fragment@1 {
        target = <&spi0>;
        __overlay__ {
            #address-cells = <1>;
            #size-cells = <0>;
            spi@0 {
                status = "okay";
                compatible = "microchip,mcp2515";
                reg = <0x0>;
                spi-max-frequency = <10000000>;
                nvidia,enable-hw-based-cs;
                nvidia,rx-clk-tap-delay = <0x7>;
                clocks = <&can_clock>;
                interrupt-parent = <&gpio>;
                interrupts = <TEGRA_GPIO(Z, 0) 0x1>;
                controller-data {
                    nvidia,cs-setup-clk-count = <0x1e>;
                    nvidia,cs-hold-clk-count = <0x1e>;
                    nvidia,rx-clk-tap-delay = <0x1f>;
                    nvidia,tx-clk-tap-delay = <0x0>;
                };
            };
            spi@1 {
                status = "disabled";
            };
        };
    };

    fragment@2 {
        target = <&spi1>;
        __overlay__ {
            #address-cells = <1>;
            #size-cells = <0>;
            spi@0 {
                status = "disabled";
            };            
            spi@1 {
                status = "okay";
                compatible = "microchip,mcp2515";
                reg = <0x0>;
                spi-max-frequency = <10000000>;
                nvidia,enable-hw-based-cs;
                nvidia,rx-clk-tap-delay = <0x7>;
                clocks = <&can_clock>;
                interrupt-parent = <&gpio>;
                interrupts = <TEGRA_GPIO(V, 0) 0x1>;
                controller-data {
                    nvidia,cs-setup-clk-count = <0x1e>;
                    nvidia,cs-hold-clk-count = <0x1e>;
                    nvidia,rx-clk-tap-delay = <0x1f>;
                    nvidia,tx-clk-tap-delay = <0x0>;
                };
            };
        };
    };

    fragment@3 {
        target = <&pinmux>;
        __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&jetson_io_pinmux>;

            jetson_io_pinmux: exp-header-pinmux {
                hdr40-pin19 {
                    nvidia,pins = "spi1_mosi_pc0";
                    nvidia,function = "spi1";
                    nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
                hdr40-pin21 {
                    nvidia,pins = "spi1_miso_pc1";
                    nvidia,function = "spi1";
                    nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
                hdr40-pin23 {
                    nvidia,pins = "spi1_sck_pc2";
                    nvidia,function = "spi1";
                    nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
                hdr40-pin24 {
                    nvidia,pins = "spi1_cs0_pc3";
                    nvidia,function = "spi1";
                    nvidia,pull = <TEGRA_PIN_PULL_UP>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
                hdr40-pin37 {
                    nvidia,pins = "spi2_mosi_pb4";
                    nvidia,function = "spi2";
                    nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
                hdr40-pin22 {
                    nvidia,pins = "spi2_miso_pb5";
                    nvidia,function = "spi2";
                    nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
                hdr40-pin13 {
                    nvidia,pins = "spi2_sck_pb6";
                    nvidia,function = "spi2";
                    nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
                hdr40-pin18 {
                    nvidia,pins = "spi2_cs0_pb7";
                    nvidia,function = "spi2";
                    nvidia,pull = <TEGRA_PIN_PULL_UP>;
                    nvidia,tristate = <TEGRA_PIN_DISABLE>;
                    nvidia,enable-input = <TEGRA_PIN_ENABLE>;
                };
            };
        };
    };
};
```

**Compiling and Installing:**
3. Ran `make all_jetsonnano` in the `seeed-linux-dtoverlays` directory.
4. Removed existing overlays:

```
sudo rm /boot/tegra210-p3448-0000-p3449-0000-a01-mcp251x.dtbo
sudo rm /boot/tegra210-p3448-0000-p3449-0000-a02-mcp251x.dtbo
```

5. Created `.dtbo` file:

```
sudo make install_jetsonnano
```

6. Copied the generated `jetson-mcp251x.dtbo` file to `/boot/`:

```
cd /seeed-linux-dtoverlays-master/overlays/jetsonnano
sudo cp jetson-mcp251x.dtbo /boot/
ls /boot/jetson-mcp251x.dtbo
```

7. Applied the overlay:

```
fdtoverlay -i tegra210-p3448-0000-p3449-0000-b00.dtb -o tegra210-p3448-0000-p3449-0000-b00-extended.dtb jetson-mcp251x.dtbo
```

8. Edited `extlinux.conf`:

```
cd /boot/extlinux
```

Added:

```
FDT /boot/tegra210-p3448-0000-p3449-0000-b00-extended.dtb
```

**Reboot and Verify:**
9. Rebooted the system:

```
sudo reboot
```

10. Verified SPI setup with:

```
dmesg | grep spi
```

**Issue:**
Despite following these steps, I encounter the following errors:

```
[    5.210328] mcp251x spi0.0: MCP251x didn't enter in conf mode after reset
[    5.220996] mcp251x spi0.0: Probe failed, err=16
[    5.249674] mcp251x: probe of spi0.0 failed with error -16
[    6.270321] mcp251x spi1.0: MCP251x didn't enter in conf mode after reset
[    6.277363] mcp251x spi1.0: Probe failed, err=16
[    6.285015] mcp251x: probe of spi1.0 failed with error -16
```

I was able to establish the CAN connection following the same steps before and successfully send and receive CAN messages, but now i am stuck with this error.
please let know if there is any solution

bigbearishappy, 2024-08-09:
Can you share you hardware connection between jetson nano and you CAN board?
And can you dump the device-tree of you device with the following command?
`dtc -I fs /proc/device-tree > dts.txt`

bigbearishappy, 2024-09-10:
I hope your problem have been solved. If not, you can reopen this issue~

closed 2024-09-10 by bigbearishappy