Seeed-Studio/seeed-linux-dtoverlays #112: Compile error on Ubuntu 22.04.5

rstaph, 2024-11-12:
During Compile I get the following error:

```
DTC     overlays/rpi/reComputer-AI-box-overlay.dtbo
Error: overlays/rpi/reComputer-AI-box-overlay.dts:150.12-13 syntax error
FATAL ERROR: Unable to parse input tree
make[1]: *** [Makefile:217: overlays/rpi/reComputer-AI-box-overlay.dtbo] Error 1
make: *** [Makefile:156: all_rpi] Error 2
```

Follow the directions from:  https://wiki.seeedstudio.com/install-ubuntu-on-reterminal/

On Ubuntu 22.04.5

rstaph, 2024-11-12:
Compiles but on trying to change display settings per instructions I get "Changes Cannot be Applied   This could be due to hardware limitations" on the settings window.

is-qian, 2024-11-15:
I think it's because there was a problem with the naming in a dts file, which has been fixed in this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/dfe57b499b854d1c8dde656ca1f842a57ad984ae). Thank you for your attempt.

closed 2024-11-15 by is-qian