Seeed-Studio/seeed-linux-dtoverlays #113: reTerminal DM screen does not light on Bookworm

Kasperror, 2025-01-30:
After flashing official newest Raspberry Pi OS 64-bit (Bookworm) and installing the overlays like instruction says approx. 4 out of 5 times I turn on the reTerminal DM the screen does not turn on at all. Doesn't matter if I use reboot or unplug/plug the device. The behavior is random. I've got 5 of those and its a similar case on 2 which I flashed with new OS. The ones with shipped OS work normally (but I cannot use it for my product tho)

[dmsg.txt](https://github.com/user-attachments/files/18605273/dmsg.txt)

Output of `dmesg | grep -i error`:
`Goodix-TS 10-005d: Direct firmware load for goodix_9271_cfg.bin failed with error -2`
I'm not sure if its related cuz it's the same when the screen actually turns on but I attach full `dmesg` output in .txt file below

Kernel version:
`6.6.62+rpt-rpi-v8`

the `/boot/firmware/config.txt`:
```# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt. 
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required    
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]

[all]
dtparam=i2c_vc=on
enable_uart=1
dtparam=ant2
disable_splash=1
ignore_lcd=1
dtoverlay=dwc2,dr_mode=host
dtoverlay=vc4-kms-v3d-pi4
dtoverlay=i2c1,pins_2_3
dtoverlay=i2c3,pins_4_5
dtoverlay=imx219,cam0
dtoverlay=reTerminal-DM
```

Kasperror, 2025-02-01:
I checked with previous versions of OS/overlays which worked well for me [exact version here](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/71ecf931ea4e359078182266de4e389be7f114c9) and it works normally so something is wrong with recent versions of overlays.

Kasperror, 2025-02-02:
If it's if any help I attach files for comparison 

[dmesg_when_ok.txt](https://github.com/user-attachments/files/18630952/dmesg_when_ok.txt)

[dmesg_when_screen_blanks.txt](https://github.com/user-attachments/files/18630953/dmesg_when_screen_blanks.txt)

bigbearishappy, 2025-02-05:
First of all, The log you showed `Goodix-TS 10-005d: Direct firmware load for goodix_9271_cfg.bin failed with error -2`. It's a log of touch screen.It's OK if we get this log, cause we already have flash it into the touch screen chip.
And I compare the log you provided. Both of them had driver crash log. It seems that something is wrong with the LCD driver now.
We will test it and feedback to you.

Kasperror, 2025-02-06:
> First of all, The log you showed `Goodix-TS 10-005d: Direct firmware load for goodix_9271_cfg.bin failed with error -2`. It's a log of touch screen.It's OK if we get this log, cause we already have [...]

Thanks a lot, if you need any further information I'm happy to help

Cyphrz, 2025-02-08:
Can confirm, flashed latest PiOS, followed directions and no screen. Previously was functional on Bullseye.

Possibly related to the switch to labwc as the default.

is-qian, 2025-02-08:
I remember testing this under labwc before. I guess it is related to this  [commit ](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/1ae0350dc93050abfc34dbc393f6c402f2b261f3). I will verify it.

bclswl0827, 2025-02-08:
I'm also following up.

See https://github.com/Seeed-Studio/wiki-documents/discussions/182#discussioncomment-11988957

Kasperror, 2025-02-20:
Any updates on the matter? I've noticed that we used to have 100% chance of blackscreen on first reboot after installing overlays, now it almost always works on first reboot (I see there were some commits in the repo). No changes to overall chance of getting a blackscreen though.

is-qian, 2025-02-21:
Have you tried this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/7432121edcd97408a352b74bdfef7898533cf393)? I tested it yesterday, and so far, no black screen has occurred. I have turned the screen on and off about 2000 times and restarted it more than 30 times.

> Any updates on the matter? I've noticed that we used to have 100% chance of blackscreen on first reboot after installing overlays, now it almost always works on first reboot (I see there were some commits in the repo). No changes to overall chance of getting a blackscreen though.

bclswl0827, 2025-02-24:
> Any updates on the matter? I've noticed that we used to have 100% chance of blackscreen on first reboot after installing overlays, now it almost always works on first reboot (I see there were some commits in the repo). No changes to overall chance of getting a blackscreen though.

Hi, please update your OS software and libraries to the latest version
```
$ sudo apt update
$ sudo apt upgrade
```
then try to reinstall the seeed-linux-dtoverlays, now it should work

Cyphrz, 2025-03-05:
Still not working

is-qian, 2025-03-05:
What is your system version? We previously tested it on the version from February 2025. Alternatively, you can use the image provided [here](https://github.com/Seeed-Studio/pi-gen-expand).

matsujirushi, 2025-03-12:
I tried latest seeed driver and latest RasPi OS 64-bit.
A problem occurred.

### Environment

* reTerminal DM
* Raspberry OS Desktop 2024-11-19 64-bit (2024-11-19-raspios-bookworm-arm64.img.xz)
* seeed-linux-dtoverlays

  ```
  885862c (HEAD -> master, origin/master, origin/HEAD) fix pcf8563w compatible with kernel 5.15
  ```

### Procedure

1. Flash 2024-11-19-raspios-bookworm-arm64.img.xz to eMMC in reTerminal DM using Raspberry Pi Imager.
2. Upgrade packages.

    ```
    sudo apt update
    sudo apt upgrade
    sudo reboot
    ```

3. Install seeed-linux-dtoverlays.

    ```
    sudo apt install git
    git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
    cd seeed-linux-dtoverlays
    sudo ./scripts/reTerminal.sh --device reTerminal-DM
    sudo reboot
    ```

4. Repeat `sudo reboot`.

### Result

1st ... OK.
2nd ... OK.
3rd ... NG. (backlight=ON, But screen not displayed.)
4th ... NG. (backlight=ON, But screen not displayed.)
5th ... OK
6th ... OK
7th ... NG. (backlight=ON, But screen not displayed.)

### Additional Infomation

```
pi@raspberrypi:~/seeed-linux-dtoverlays $ cat /boot/issue.txt
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4
pi@raspberrypi:~/seeed-linux-dtoverlays $ uname -r
6.6.74+rpt-rpi-v8
pi@raspberrypi:~/seeed-linux-dtoverlays $ uname -m
aarch64
pi@raspberrypi:~/seeed-linux-dtoverlays $ dpkg --print-architecture
arm64
pi@raspberrypi:~/seeed-linux-dtoverlays $ getconf LONG_BIT
64
```

is-qian, 2025-03-12:
> I tried latest seeed driver and latest RasPi OS 64-bit.
> A problem occurred.
>
> ### Environment
>
> * reTerminal DM
> * Raspberry OS Desktop 2024-11-19 64-bit (2024-11-19-raspios-bookworm-arm64.img.xz)
> * [...]

Can you tell me what command you use to turn on and off the screen?

matsujirushi, 2025-03-12:
Hi,

> Can you tell me what command you use to turn on and off the screen?

OS reboot. `sudo reboot`

It is not just on/off for screen.

is-qian, 2025-03-14:
I think it’s because the new system uses initramfs for booting. Regenerating the initramfs file after executing the script to install the driver seems to resolve this issue. 
```
sudo update-initramfs -c -k $(uname -r)
```

Kasperror, 2025-03-26:
> I think it’s because the new system uses initramfs for booting. Regenerating the initramfs file after executing the script to install the driver seems to resolve this issue.
> 
> ```
> sudo update-initramfs -c -k $(uname -r)
> ```

I don't want to steal your shine, can you please PR this? - it works for me, tested on 10 boots on 4 different machines

Cyphrz, 2025-03-30:
For me at least, downloading the latest, and running, even doing `sudo update-initramfs -c -k $(uname -r)` after doesn't help. Will try a full clean reimage.

Kasperror, 2025-03-30:
> For me at least, downloading the latest, and running, even doing `sudo update-initramfs -c -k $(uname -r)` after doesn't help. Will try a full clean reimage.

Are you installing full newest PiOS (Bookworm with desktop environment). I can confirm that it works for 15 devices I've set up but only if its installed after first reboot after the `reTerminal` setup script as I've mentioned.

matsujirushi, 2025-04-01:
Hi @Olaruci ,
I created new issue.
Please watch the issue #123 .

matsujirushi, 2025-04-01:
Hi @Olaruci ,
Thanks for the information.
The problem has been resolved.
Please check the issue #123 .
If that doesn't solve your problem, please create a new issue. 🙏

matsujirushi, 2025-04-01:
> > I think it’s because the new system uses initramfs for booting. Regenerating the initramfs file after executing the script to install the driver seems to resolve this issue.
> > ```
> > sudo update-initramfs -c -k $(uname -r)
> > ```
> 
> I see that line has been added to the codebase BUT it doesn't seem to work either. The solution works but only when ran after whole process. Maybe the initramfs is changed somewhere further than the install script?
> 
> Edit: even squeezing it right after the install script won't work the `initramfs` command has to be run after first reboot after the `reTerminal` setup script.

I think this has been an inherent problem with reTerminal.sh for some time.
I am upgrading the Kernel inside reTerminal.sh, but it will not change until I reboot the OS, so some commands do not return new OS information. (e.g. uname -r)
Therefore, I `sudo apt update && sudo apt upgrade` before running reTermina.sh.
(The `--keep-kernel` option might work, but I don't use it.)

closed 2025-04-23 by github-actions[bot]