Seeed-Studio/seeed-linux-dtoverlays #114: LTE module no longer works after installing newest overlays

Kasperror, 2025-02-01:
After flashing newest Bookworm before I install overlays I can see `usb0` interface in `ifconfig`. 
After installing the overlays according to the instructions I configure the EC25 PCIe module in `minicom` but I can't see `usb0` no more - now it shows up as `eth1`.
When i disconnect the ethernet cable I want to use the LTE as internet source
`ping -I eth1 8.8.8.8` works correctly (i can see packets being transferred) but
`ping -I eth1 google.com` results in unresolved DNS. 
There is no way to setup DNS for `eth1` as in newest version everything is controlled by `nmcli` which does not recognize either `eth1` nor `usb0`. It just shows `ttyUSB2: disconnected` as one of the devices.

Please provide some instructions and solution to actually use your product now.

bigbearishappy, 2025-02-05:
Please tell us which device you used, we will test it and feedback to you.

Kasperror, 2025-02-06:
> Please tell us which device you used, we will test it and feedback to you.

it's reTerminal DM with EC25-EUX  module (ordered as package with reTerminal DMs).

bigbearishappy, 2025-02-11:
Have you check the kernel config of PPP? Maybe it's not enabled.

closed 2025-03-06 by github-actions[bot]