Seeed-Studio/seeed-linux-dtoverlays #116: Display sleep: DSI transfer failed, resetting: -110

franzos, 2025-02-28:
When the display goes to "sleep" it is unable to wake-up after some time.

```bash
[ 2219.387747] mipi_dsi: Disable panel
[ 2219.390998] mipi_dsi: Unprepare panel
[ 2277.800231] mipi_dsi: Prepare panel
[ 2277.856597] mipi_dsi: Prepare ILI9881D
[ 2278.876621] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[ 2278.876645] vc4_dsi fe700000.dsi: instat: 0x00000000
[ 2278.876655] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[ 2278.876728] mipi_dsi: (ili9881d_prepare) No LCD connected,pls check your hardware!
[ 2278.881515] mipi_dsi: Enable panel
[ 2878.906422] mipi_dsi: Disable panel
[ 2878.909430] mipi_dsi 1-0045: Skipping unprepare of already unprepared panel
[ 4410.507456] mipi_dsi: Prepare panel
[ 4410.567928] mipi_dsi: Prepare ILI9881D
[ 4410.569819] [drm:vc4_dsi_irq_handler [vc4]] *ERROR* DSI1: turnaround timeout error
[ 4411.575946] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[ 4411.575961] vc4_dsi fe700000.dsi: instat: 0x00000000
[ 4411.575968] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[ 4411.576021] mipi_dsi: (ili9881d_prepare) No LCD connected,pls check your hardware!
[ 4411.582745] mipi_dsi: Enable panel
```

Display

```bash
$ DISPLAY=:0.0 xrandr
Screen 0: minimum 320 x 200, current 1280 x 720, maximum 7680 x 7680
HDMI-1 disconnected primary (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
DSI-1 connected 1280x720+0+0 right (normal left inverted right x axis y axis) 0mm x 0mm
   720x1280      60.00*+
```

DPMS

```bash
$ DISPLAY=:0.0 xset q
Keyboard Control:
  auto repeat:  on    key click percent:  0    LED mask:  00000000
  XKB indicators:
    00: Caps Lock:   off    01: Num Lock:    off    02: Scroll Lock: off
    03: Compose:     off    04: Kana:        off    05: Sleep:       off
    06: Suspend:     off    07: Mute:        off    08: Misc:        off
    09: Mail:        off    10: Charging:    off    11: Shift Lock:  off
    12: Group 2:     off    13: Mouse Keys:  off
  auto repeat delay:  660    repeat rate:  25
  auto repeating keys:  00ffffffdffffbbf
                        fadfffefffedffff
                        9fffffffffffffff
                        fff7ffffffffffff
  bell percent:  50    bell pitch:  400    bell duration:  100
Pointer Control:
  acceleration:  2/1    threshold:  4
Screen Saver:
  prefer blanking:  yes    allow exposures:  yes
  timeout:  600    cycle:  600
Colors:
  default colormap:  0x20    BlackPixel:  0x0    WhitePixel:  0xffffff
Font Path:
  /usr/share/fonts/X11/Type1,built-ins
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Enabled
  Monitor is On
```

**To Reproduce**:

- Enable sleep `xset +dpms s on`
- Wait until display goes to sleep; and some more

**Expected behavior**:

Display wakes up from sleep.

**Additional context**:

- Using [/2024-03-14-Raspbian-reTerminal/2024-03-14-Raspbian-reTerminal-arm64.zip](https://files.seeedstudio.com/wiki/ReTerminal/RPiOS_Images/2024-03-14-Raspbian-reTerminal/2024-03-14-Raspbian-reTerminal-arm64.zip)
- No updates

```
Linux vhh.gateway-VANILLA 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
```

is-qian, 2025-03-04:
The previous image seems to have this [issue](https://github.com/raspberrypi/linux/issues/4323). You can try using the latest [img](https://github.com/Seeed-Studio/pi-gen-expand).

is-qian, 2025-03-07:
Thank you for your reminder. I will pass your suggestion on to our colleagues in charge of the wiki.