Seeed-Studio/seeed-linux-dtoverlays #117: [reTerminal DM] Backlight does not turn on

matsujirushi, 2025-03-04:
### Equipment

* reTerminal DM
* Ubuntu Desktop 24.04.2 64bit (ubuntu-24.04.2-preinstalled-desktop-arm64+raspi.img.xz)
* seeed-linux-dtoverlays ( https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/0b1c58482e84f2c6e238d03d66e330f5f5d5bc2b )

### Procedure

1. Flash ubuntu-24.04.2-preinstalled-desktop-arm64+raspi.img.xz to eMMC in reTerminal DM using Raspberry Pi Imager.
2. Upgrade packages.
    ```
    sudo apt update
    sudo apt upgrade
    sudo reboot
    ```
3. Install seeed-linux-dtoverlays.
    ```
    sudo apt install git
    git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
    cd seeed-linux-dtoverlays
    sudo ./scripts/reTerminal.sh --device reTerminal-DM
    sudo reboot
    ```

**Screen is not displayed.**

### Logs

[installdriver.txt](https://github.com/user-attachments/files/19079068/installdriver.txt)

[dmesg.txt](https://github.com/user-attachments/files/19079070/dmesg.txt)

### Additional Information

`echo 0 > /sys/class/backlight/lcd_backlight/bl_power` command will bring up a screen.

```
$ cat /sys/class/backlight/lcd_backlight/bl_power
4
$ sudo -i
# echo 0 > /sys/class/backlight/lcd_backlight/bl_power
# exit
$
```

is-qian, 2025-03-05:
Before you execute apt upgrade, can the screen be normally turned on? I guess it's also related to the kernel update, similar to this [issue](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/113). I will go and confirm it on my device.

matsujirushi, 2025-03-05:
Hi @is-qian ,

> Before you execute apt upgrade, can the screen be normally turned on?

No, the screen is not displayed because the timing is such that the Seeed driver has not been installed after installing Ubuntu. 😉

Thank you for your support.
I will wait to try it on your device.

is-qian, 2025-03-06:
Thanks for the context! I think the issue stems from [this](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/include/drm/drm_panel.h#L331).In my Ubuntu setup, I’ve read the configuration, and it seems that drm_panel_of_backlight only returns a non-zero value when CONFIG_DRM_PANEL=y and CONFIG_BACKLIGHT_CLASS_DEVICE=y, or when CONFIG_DRM=m and CONFIG_BACKLIGHT_CLASS_DEVICE=y. To fundamentally resolve this, I’d need to reconfigure and recompile the kernel. However, I’ve also tried bypassing DRM by directly acquiring the backlight in the display driver, which works fine too. Below are my changes—I’ve tested them, and the backlight turns on normally.
```
diff --git a/modules/ili9881d/ili9881d.c b/modules/ili9881d/ili9881d.c
index 0c88832..bc8f926 100644
--- a/modules/ili9881d/ili9881d.c
+++ b/modules/ili9881d/ili9881d.c
@@ -534,10 +534,11 @@ static int ili9881d_dsi_probe(struct mipi_dsi_device *dsi)
                return ret;
        }

-       ret = drm_panel_of_backlight(&ctx->panel);
-       if (ret)
-               return ret;
+       // ret = drm_panel_of_backlight(&ctx->panel);
+       // if (ret)
+       //      return ret;

+       ctx->panel.backlight = devm_of_find_backlight(ctx->panel.dev);
        drm_panel_add(&ctx->panel);

        dsi->mode_flags = ctx->desc->mode_flags;
```
Let me know if there’s anything else I can refine!

is-qian, 2025-03-06:
Here’s the configuration I retrieved from my Ubuntu system:
```
$ cat /boot/config-6.8.0-1020-raspi | grep CONFIG_DRM_PANEL=
CONFIG_DRM_PANEL=y
$ cat /boot/config-6.8.0-1020-raspi | grep CONFIG_DRM=                                 
CONFIG_DRM=y
$ cat /boot/config-6.8.0-1020-raspi | grep CONFIG_BACKLIGHT_CLASS_DEVICE=
CONFIG_BACKLIGHT_CLASS_DEVICE=m
```

matsujirushi, 2025-03-07:
Hi @is-qian ,
Thanks for the great information!

> I’d need to reconfigure and recompile the kernel.

It is hard to get users to kernel recompile, so I would like to address this issue by changing the driver.

I have confirmed that the screen can be displayed on my reTerminal DM with the driver change. 👍
However, there are two points of concern.

1. Can compatibility be maintained? (Possible with definition macros?)
1. Screen lock function does not return to the touch screen (can return with mouse).

Can this be addressed by changing the driver?

### Additional Information

* Ubuntu 24.04.2 64bit
  * 6.8.0-1020-raspi
    ```
    $ cat /boot/config-6.8.0-1020-raspi | egrep 'CONFIG_DRM_PANEL=|CONFIG_DRM=|CONFIG_BACKLIGHT_CLASS_DEVICE='
    CONFIG_DRM=y
    CONFIG_DRM_PANEL=y
    CONFIG_BACKLIGHT_CLASS_DEVICE=m
    ```
* Ubuntu 22.04.5 64bit
  * 5.15.0-1073-raspi
    ```
    $ cat /boot/config-5.15.0-1073-raspi | egrep 'CONFIG_DRM_PANEL=|CONFIG_DRM=|CONFIG_BACKLIGHT_CLASS_DEVICE='
    CONFIG_DRM=m
    CONFIG_DRM_PANEL=y
    CONFIG_BACKLIGHT_CLASS_DEVICE=y
    ```
* RasPiOS bookworm 2024/11/19 64bit
  * 6.6.74+rpt-rpi-v8
    ```
    $ cat /boot/config-6.6.74+rpt-rpi-v8 | egrep 'CONFIG_DRM_PANEL=|CONFIG_DRM=|CONFIG_BACKLIGHT_CLASS_DEVICE='
    CONFIG_DRM=m
    CONFIG_DRM_PANEL=y
    CONFIG_BACKLIGHT_CLASS_DEVICE=m
    ```

is-qian, 2025-03-11:
Compatibility has been made in this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/beec2c0cc6ac986285d3c408b32fe932d9a5fd83).
The issue of the touchscreen not waking the screen stems from the Goodix touchscreen driver not marking the device as a wakeup source. In Ubuntu's default Wayland environment, screen wakeup relies on the device being properly marked as a wakeup source (e.g., via device_init_wakeup). Switching to X11 can resolve this issue by editing /etc/gdm3/custom.conf to set WaylandEnable=false and restarting GDM with sudo systemctl restart gdm3. This works because X11, unlike Wayland, does not strictly depend on the wakeup source marking; instead, it directly monitors events from /dev/input/* and treats any input event as an activity signal to wake the screen, whereas Wayland (via Mutter and libinput) relies more heavily on the kernel's power management mechanisms.

is-qian, 2025-03-11:
The issue of the touchscreen not waking the screen stems from the [Goodix touchscreen driver](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/input/touchscreen/goodix.c) not marking the device as a wakeup source. In Ubuntu's default Wayland environment, screen wakeup relies on the device being properly marked as a wakeup source (e.g., via device_init_wakeup). Switching to X11 can resolve this issue by editing /etc/gdm3/custom.conf to set WaylandEnable=false and restarting GDM with sudo systemctl restart gdm3. This works because X11, unlike Wayland, does not strictly depend on the wakeup source marking; instead, it directly monitors events from /dev/input/* and treats any input event as an activity signal to wake the screen, whereas Wayland (via Mutter and libinput) relies more heavily on the kernel's power management mechanisms.

matsujirushi, 2025-03-12:
Hi @is-qian ,
Thanks for support!

Screen is displayed. 🙌

> Screen lock function does not return to the touch screen (can return with mouse).

I will create a new issue. 😉

closed 2025-03-12 by matsujirushi