Seeed-Studio/seeed-linux-dtoverlays #118: [reTerminal DM] no member named ‘get_orientation’

matsujirushi, 2025-03-07:
### Equipment

* reTerminal DM
* Ubuntu Desktop 22.04.5 64bit (ubuntu-22.04.5-preinstalled-desktop-arm64+raspi.img.xz)
* seeed-linux-dtoverlays ( https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/0b1c58482e84f2c6e238d03d66e330f5f5d5bc2b )

### Procedure

1. Flash ubuntu-22.04.5-preinstalled-desktop-arm64+raspi.img.xz to eMMC in reTerminal DM using Raspberry Pi Imager.
2. Upgrade packages.
    ```
    sudo apt update
    sudo apt upgrade
    sudo reboot
    ```
3. Install seeed-linux-dtoverlays.
    ```
    sudo apt install git
    git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
    cd seeed-linux-dtoverlays
    sudo ./scripts/reTerminal.sh --device reTerminal-DM
    ```

I get a compilation error.

```
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-1073-raspi'
  CC [M]  /var/lib/dkms/ili9881d/0.1/build/ili9881d.o
/var/lib/dkms/ili9881d/0.1/build/ili9881d.c:495:10: error: ‘const struct drm_panel_funcs’ has no member named ‘get_orientation’
  495 |         .get_orientation = ili9881d_get_orientation,
      |          ^~~~~~~~~~~~~~~
/var/lib/dkms/ili9881d/0.1/build/ili9881d.c:495:28: error: initialization of ‘int (*)(struct drm_panel *, unsigned int,  struct display_timing *)’ from incompatible pointer type ‘enum drm_panel_orientation (*)(struct drm_panel *)’ [-Werror=incompatible-pointer-types]
  495 |         .get_orientation = ili9881d_get_orientation,
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
```

is-qian, 2025-03-07:
It's possible that get_orientation is not available in 5.15, the screen driver might need some adaptation.

is-qian, 2025-03-11:
In this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/beec2c0cc6ac986285d3c408b32fe932d9a5fd83), the issue has been fixed.

closed 2025-03-11 by is-qian