Seeed-Studio/seeed-linux-dtoverlays #120: [reTerminal DM] error: implicit declaration of function ‘rtc_register_device’

matsujirushi, 2025-03-11:
A different error is occurring in this environment #118 . 😣
Could you please check?

Error message:

```
  CC [M]  /var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.o
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c: In function ‘pcf8563_probe’:
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c:761:11: error: implicit declaration of function ‘rtc_register_device’; did you mean ‘i2c_unregister_device’? [-Werror=implicit-function-declaration]
  761 |     err = rtc_register_device(pcf8563->rtc);
      |           ^~~~~~~~~~~~~~~~~~~
      |           i2c_unregister_device
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c: At top level:
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c:795:14: error: initialization of ‘int (*)(struct i2c_client *, const struct i2c_device_id *)’ from incompatible pointer type ‘int (*)(struct i2c_client *)’ [-Werror=incompatible-pointer-types]
  795 |     .probe = pcf8563_probe,
      |              ^~~~~~~~~~~~~
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c:795:14: note: (near initialization for ‘pcf8563_driver.probe’)
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c:538:12: warning: ‘pcf8563_irq_enable’ defined but not used [-Wunused-function]
  538 | static int pcf8563_irq_enable(struct device *dev, unsigned int enabled)
      |            ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c:520:12: warning: ‘pcf8563_rtc_set_alarm’ defined but not used [-Wunused-function]
  520 | static int pcf8563_rtc_set_alarm(struct device *dev, struct rtc_wkalrm *tm)
      |            ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c:487:12: warning: ‘pcf8563_rtc_read_alarm’ defined but not used [-Wunused-function]
  487 | static int pcf8563_rtc_read_alarm(struct device *dev, struct rtc_wkalrm *tm)
      |            ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.c:366:20: warning: ‘pcf8563_irq’ defined but not used [-Wunused-function]
  366 | static irqreturn_t pcf8563_irq(int irq, void *dev_id)
      |                    ^~~~~~~~~~~
cc1: some warnings being treated as errors
```

Used seeed-linux-dtoverlays:

```
7013b3e (HEAD -> master, origin/master, origin/HEAD) Merge pull request #119 from Seeed-Studio/fix-drm
```

is-qian, 2025-03-11:
Yes, I found this issue too and I fixed it in another [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/8b5ee0fe5f042645fbea6ea0610fcea324484753). You can try it again in the latest code.

matsujirushi, 2025-03-11:
Hi @is-qian ,
[The commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/8b5ee0fe5f042645fbea6ea0610fcea324484753) is already included.

```
$ git log --oneline
7013b3e (HEAD -> master, origin/master, origin/HEAD) Merge pull request #119 from Seeed-Studio/fix-drm
8b5ee0f fix pcf8563w compatible with kernel 5.15   <---
beec2c0 ili9881:Fixed the issue where backlight fails to illuminate in Ubuntu 6.8.
0b1c584 DM:Synchronize and update the ili9881 driver.
7432121 DM:Update ili9881 driver to fix loading error in new GPU/DRM.
```

The kernel version in my Ubuntu 22.04 is 5.15.

```
$ uname -r
5.15.0-1073-raspi
```

There may be a problem with the decision conditions.

https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/master/modules/rtc-pcf8563w/rtc-pcf8563w.c#L706

is-qian, 2025-03-12:
Sorry, there is a problem with my commit. Please apply the [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/885862cce068feb29a9e021269c2659ba6890220) and try again. I apologize again for my negligence.

matsujirushi, 2025-03-12:
Hi @is-qian ,

The driver was properly compiled and installed, and a screen was displayed.

Thanks for support!

closed 2025-03-12 by matsujirushi