Seeed-Studio/seeed-linux-dtoverlays #123: [reTerminal DM] fatal error: asm/unaligned.h: No such file or directory

matsujirushi, 2025-04-01:
> Hi , I also get a dark screen after install of latest Raspberry OS 64bit ,  kernel 6.12.20.
> It seems like the drivers installation is not finalised. I get this error at the end of the ` sudo ./scripts/reTerminal.sh --device reTerminal-DM` command:
> 
> 
> 
> `Building module:`
> `Cleaning build area...`
> `make -j4 KERNELRELEASE=6.12.20+rpt-rpi-v8 all KVERSION=6.12.20+rpt-rpi-v8....(bad exit status: 2)`
> `Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-v8 (aarch64)`
> `Consult /var/lib/dkms/ch34x/0.1/build/make.log for more information.`
> `Can't compile with this kernel, aborting`
> `Please try to compile with the option --compat-kernel`
> `DKMS make.log for ch34x-0.1 for kernel 6.12.20+rpt-rpi-v8 (aarch64)`
> `Mon 31 Mar 21:24:13 EEST 2025`
> `make -C /lib/modules/6.12.20+rpt-rpi-v8/build M=/var/lib/dkms/ch34x/0.1/build modules`
> `make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'`
>  ` CC [M]  /var/lib/dkms/ch34x/0.1/build/ch343.o`
> `/var/lib/dkms/ch34x/0.1/build/ch343.c:48:10: fatal error: asm/unaligned.h: No such file or directory
>    48 | #include <asm/unaligned.h>
>       |          ^~~~~~~~~~~~~~~~~
> compilation terminated.`
> `make[3]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: ` `/var/lib/dkms/ch34x/0.1/build/ch343.o] Error 1`
> `make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/ch34x/0.1/build] Error 2`
> `make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2`
> `make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'`
> `make: *** [Makefile:13: all] Error 2`
> 
> Tried this:
> ` sudo rmmod ili9881 `
> `rmmod: ERROR: Module ili9881d is not currently loaded `
> 
> And this with no change after reboot: 
> `sudo update-initramfs -c -k $(uname -r)`
> `update-initramfs: Generating /boot/initrd.img-6.12.20+rpt-rpi-v8`
> `'/boot/initrd.img-6.12.20+rpt-rpi-v8' -> '/boot/firmware/initramfs8'
> ` 

 _Originally posted by @Olaruci in [#113](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/113#issuecomment-2767185287)_

matsujirushi, 2025-04-01:
### Environment

* reTerminal DM
* Raspberry OS Desktop 2024-11-19 64-bit (2024-11-19-raspios-bookworm-arm64.img.xz)
* seeed-linux-dtoverlays (04cd399)

### Procedure

1. Flash 2024-11-19-raspios-bookworm-arm64.img.xz to eMMC in reTerminal DM using Raspberry Pi Imager.
2. Upgrade packages.

    ```
    sudo apt update
    sudo apt upgrade
    sudo reboot
    ```

3. Install seeed-linux-dtoverlays.

    ```
    sudo apt install git
    git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
    cd seeed-linux-dtoverlays
    sudo ./scripts/reTerminal.sh --device reTerminal-DM
    ```

### Result

```
Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.12.20+rpt-rpi-v8 all KVERSION=6.12.20+rpt-rpi-v8...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/ch34x/0.1/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
DKMS make.log for ch34x-0.1 for kernel 6.12.20+rpt-rpi-v8 (aarch64)
Tue  1 Apr 10:04:56 JST 2025
make -C /lib/modules/6.12.20+rpt-rpi-v8/build M=/var/lib/dkms/ch34x/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/ch34x/0.1/build/ch343.o
/var/lib/dkms/ch34x/0.1/build/ch343.c:48:10: fatal error: asm/unaligned.h: No such file or directory
   48 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /var/lib/dkms/ch34x/0.1/build/ch343.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/ch34x/0.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
make: *** [Makefile:13: all] Error 2
```

Kernel version is `6.12.20+rpt-rpi-v8`.

matsujirushi, 2025-04-01:
FYI
`move asm/unaligned.h to linux/unaligned.h`
https://github.com/raspberrypi/linux/commit/5f60d5f6bbc12e782fac78110b0ee62698f3b576

is-qian, 2025-04-01:
Thank you for your help. I have added support for ch34x on kernel 6.12 [here ](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/efd2ae4cbbb8e4fbd2c4c5f2be0e1cf5b3dd5e36).

closed 2025-04-01 by matsujirushi

Olaruci, 2025-04-01:
Yes it is fixed.  
I had to 'sudo update-initramfs -c -k $(uname -r) ' because it had some back-lit on but no image .
And also to rotate right the screen orientation but now is working perfectly after a few reboots and shutdown.

Thanks for quick response.

Olaruci, 2025-04-02:
Quick question. Is there a way to use brightness sensor of the terminal DM with Raspberry OS ?

matsujirushi, 2025-04-03:
> [@matsujirushi](https://github.com/matsujirushi) Quick question please. Is there a way to use brightness sensor of the terminal DM with Raspberry OS ?

I don't know.
I think it is better to ask the community via [forum](https://forum.seeedstudio.com/) or [discord](https://discord.com/invite/eWkprNDMU7).