Seeed-Studio/seeed-linux-dtoverlays #124: MIPI module build failure after update and fresh install

ThomasPips, 2025-04-05:
**Describe the bug**
After a full update of the Raspbian OS the kernel modules failed to build.
Made a complete reinstall of Raspbian but after this the problem persists.

Log output;

[image]

I've tried multiple searches on the internet but no resolution was found.

matsujirushi, 2025-04-05:
Hi @ThomasPips ,
Could you give me information the device (reTerminal?) and Raspberry Pi OS (2024-11-19) you are using?

matsujirushi, 2025-04-06:
I was able to reproduce it.

### Environment

* reTerminal DM
* Raspberry OS Desktop 2024-11-19 64-bit (2024-11-19-raspios-bookworm-arm64.img.xz)
* seeed-linux-dtoverlays (https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/efd2ae4cbbb8e4fbd2c4c5f2be0e1cf5b3dd5e36)

### Procedure

1. Flash 2024-11-19-raspios-bookworm-arm64.img.xz to eMMC in reTerminal using Raspberry Pi Imager.
2. Upgrade packages.

    ```
    sudo apt update
    sudo apt upgrade
    sudo reboot
    ```

3. Install seeed-linux-dtoverlays.

    ```
    sudo apt install git
    git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
    cd seeed-linux-dtoverlays
    sudo ./scripts/reTerminal.sh
    ```

### Result

```
DKMS make.log for mipi_dsi-0.1 for kernel 6.12.20+rpt-rpi-v8 (aarch64)
Sun  6 Apr 09:37:00 JST 2025
make -C /lib/modules/6.12.20+rpt-rpi-v8/build M=/var/lib/dkms/mipi_dsi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/touch_panel.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: In function ‘backlight_update’:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:305:35: error: implicit declaration of function ‘bl_get_data’; did you mean ‘acpi_get_data’? [-Werror=implicit-function-declaration]
  305 |         struct i2c_mipi_dsi *md = bl_get_data(bd);
      |                                   ^~~~~~~~~~~
      |                                   acpi_get_data
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:305:35: warning: initialization of ‘struct i2c_mipi_dsi *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:306:28: error: invalid use of undefined type ‘struct backlight_device’
  306 |         int brightness = bd->props.brightness;
      |                            ^~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:308:15: error: invalid use of undefined type ‘struct backlight_device’
  308 |         if (bd->props.power != FB_BLANK_UNBLANK ||
      |               ^~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:309:19: error: invalid use of undefined type ‘struct backlight_device’
  309 |                 bd->props.fb_blank != FB_BLANK_UNBLANK ||
      |                   ^~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:310:19: error: invalid use of undefined type ‘struct backlight_device’
  310 |                 bd->props.state & (BL_CORE_SUSPENDED | BL_CORE_FBBLANK)) {
      |                   ^~
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c: In function ‘ili9881d_prepare’:
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:61:13: warning: unused variable ‘val’ [-Wunused-variable]
   61 |         u32 val[4] = {0};
      |             ^~~
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:60:13: warning: unused variable ‘addr’ [-Wunused-variable]
   60 |         u16 addr = 0xda;
      |             ^~~~
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:59:13: warning: unused variable ‘ret’ [-Wunused-variable]
   59 |         int ret = 0;
      |             ^~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:310:36: error: ‘BL_CORE_SUSPENDED’ undeclared (first use in this function); did you mean ‘DQUOT_SUSPENDED ’?
  310 |                 bd->props.state & (BL_CORE_SUSPENDED | BL_CORE_FBBLANK)) {
      |                                    ^~~~~~~~~~~~~~~~~
      |                                    DQUOT_SUSPENDED
[... 37 lines cut ...]
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:345:9: error: implicit declaration of function ‘backlight_update_status’; did you mean ‘backlight_update’? [-Werror=implicit-function-declaration]
  345 |         backlight_update_status(bd);
      |         ^~~~~~~~~~~~~~~~~~~~~~~
      |         backlight_update
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:329:37: warning: unused variable ‘props’ [-Wunused-variable]
  329 |         struct backlight_properties props;
      |                                     ^~~~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: In function ‘i2c_md_init’:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:515:9: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  515 |         if (ret < 0)
      |         ^~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:517:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  517 |                 return ret;
      |                 ^~~~~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: At top level:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:321:35: error: storage size of ‘backlight_ops’ isn’t known
  321 | static const struct backlight_ops backlight_ops = {
      |                                   ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

```
$ uname -r
6.12.20+rpt-rpi-v8
```

matsujirushi, 2025-04-06:
Note.
`linux/backlight.h` is no longer included from `linux/fb.h`.

https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/efd2ae4cbbb8e4fbd2c4c5f2be0e1cf5b3dd5e36/modules/mipi_dsi/mipi_dsi.h#L22

https://github.com/raspberrypi/linux/commit/11b4eedfc87de394ed8cc54dea87c745d37ff9dc

hoveywankenobi, 2025-04-08:
Patches to Drivers to fix this issue have been made here: [Issue 125](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/pull/125). 

It fixes the build issue however now we have a hang on reboot condition. reTerminal no longer successfully powercycles on it's own with `sudo reboot` command. Instead must be unplugged to reboot.

hoveywankenobi, 2025-04-08:
@ThomasPips  - I can confirm that the behavior after this update is consistent on both the RPi Bookworm OS built here:https://github.com/Seeed-Studio/pi-gen-expand/releases/tag/v1.1.1
as well as Ubuntu Server and Desktop 24.04.2LTS after running the latest updates for the seeed-linux-dtoverlays. 

Previously rebooting worked on both platforms but the latest kernel update forces use of the latest seeed-linux-dtoverlays. 

@is-qian @Lesords - this is an urgent issue.

hoveywankenobi, 2025-04-08:
Attaching journalctl output for one boot & reboot cycle. Hang happens after SigTerm is sent. @is-qian 

[journal-out.txt](https://github.com/user-attachments/files/19651657/journal-out.txt)

hoveywankenobi, 2025-04-08:
Have made a few more attempts at debugging what is happening here. 
Here is a jourlnalctl output with full logging turned on: 

[journal-out-verbosedebug.txt](https://github.com/user-attachments/files/19652487/journal-out-verbosedebug.txt)

Additionally in trying to isolate what driver is causing the hang I temporarily blacklisted the mipi-dsi driver using the following commands:
`echo "blacklist mipi_dsi" | sudo tee /etc/modprobe.d/blacklist-mipi-dsi.conf
sudo update-initramfs -u`

Obviously this causes the screen not to function, however the reTerminal returns to normal reboot functionality when the driver is blacklisted, so the issue definitely seems to lie in how this driver is defined. 

journalctl output while mipi is blacklisted:

[journal-out-verbosedebug-mipiBlacklisted.txt](https://github.com/user-attachments/files/19652584/journal-out-verbosedebug-mipiBlacklisted.txt)

is-qian, 2025-04-09:
Thanks for your feedback, I will look into this issue right away.

hoveywankenobi, 2025-04-09:
@is-qian - I've done some further testing and I think this comes down to some sort of race condition on the ic2_md_remove and ic2_md_shutdown functions in the mipi driver code.  I noticed that trying to do a `sudo modprobe -r mipi_dsi` from the running system ends with a segmentation fault:

```
Apr 08 19:53:18.984912 vf-seeed-base kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000638
...
Apr 08 19:53:19.590022 vf-seeed-base kernel: pc : vc4_hvs_unbind+0x20/0x160 [vc4]
Apr 08 19:53:19.590074 vf-seeed-base kernel: lr : component_unbind+0x40/0x70
...
Apr 08 19:53:19.591092 vf-seeed-base kernel:  vc4_dsi_host_detach+0x30/0x58 [vc4]
Apr 08 19:53:19.591142 vf-seeed-base kernel:  mipi_dsi_detach+0x40/0x68
Apr 08 19:53:19.591190 vf-seeed-base kernel:  i2c_md_remove+0x6c/0xa8 [mipi_dsi]
```

Where: The crash occurs in vc4_hvs_unbind (part of the VC4 DRM driver), called during mipi_dsi_detach from i2c_md_remove.
Why: A NULL pointer dereference at offset 0x638 suggests vc4_hvs_unbind accessed a field in a structure that wasn’t initialized or was already freed.
Timing: The crash happens ~150ms after "Unprepare panel" (18.832633 to 18.984912), during DSI detach.

Call Stack
```
vc4_hvs_unbind+0x20/0x160 [vc4]
component_unbind+0x40/0x70
component_unbind_all+0xd0/0xe8
vc4_component_unbind_all+0x20/0x38 [vc4]
devm_action_release+0x1c/0x30
release_nodes+0x70/0x100
devres_release_group+0xd4/0x150
component_del+0xb8/0x170
vc4_dsi_host_detach+0x30/0x58 [vc4]
mipi_dsi_detach+0x40/0x68
i2c_md_remove+0x6c/0xa8 [mipi_dsi]
```
Path: i2c_md_remove calls mipi_dsi_detach(md->dsi), which triggers vc4_dsi_host_detach. This unlinks the DSI host, leading to vc4_hvs_unbind crashing.

Diagnosis
Root Cause: The segfault in vc4_hvs_unbind suggests the VC4 driver expects a valid struct vc4_hvs or related data, but it’s NULL or invalid by the time mipi_dsi_detach calls into it. This could happen if:
Order Issue: i2c_md_remove detaches the DSI device before the DRM stack fully releases it.
Double Free: Something (e.g., devm_) already freed a resource that vc4_hvs_unbind needs.
Race: labwc or Xwayland is still using the DRM device, and the detach races with their cleanup.

Hypothesis: mipi_dsi_detach triggers vc4_dsi_host_detach, which assumes the HVS (Hardware Video Scaler) is still bound, but it’s not.

[fullLog.txt](https://github.com/user-attachments/files/19659013/fullLog.txt)

is-qian, 2025-04-09:
I think your guess is correct. After executing the reboot, I did indeed see the MIPI crash log through the serial port. Moreover, after I commented out the mipi_dsi_detach call in the driver, the error disappeared. I will continue to monitor this issue.

hoveywankenobi, 2025-04-10:
@is-qian - I can confirm that commenting out the mipi_dsi_detach does remedy the symptom of the hang during the reboot. I think the next logical question is what is causing the deadlock or contention in that function. When I added traces to the code I did find one hang in the tp_deinit function during the call to `input_unregister_device(md->input);` however skipping that just resulted in a second hang during the call to panel_disable. Seems like something is not releasing during the whole detach process.

is-qian, 2025-04-15:
I ultimately decided to remove mipi_dsi_detach(md->dsi);. After multiple tests, this resolved the issue of being unable to reboot. I referred to some drivers [here](https://github.com/raspberrypi/linux/tree/rpi-6.12.y/drivers/gpu/drm/panel), but none of them directly solved the problem. Understanding the root cause might take a long time, so for now, I have removed mipi_dsi_detach(md->dsi).

closed 2025-05-07 by github-actions[bot]