Seeed-Studio/seeed-linux-dtoverlays #127: Kernel panic when reboot or shutdown on reTerminal with latest Raspberry Pi OS Bookworm

asmagill, 2025-04-10:
Following the instructions at https://wiki.seeedstudio.com/reTerminal/#flash-raspberry-pi-os-64-bit-ubuntu-os-or-other-os-to-emmc, I have installed the latest Raspberry Pi OS, and upgraded it. Everything seems to work, except when I reboot or shut the device down, a kernel panic occurs.

~~~
[  535.084458] gpio-keys gpio_keys: failed to configure IRQ 40 as w
akeup source: -22                                                               
[  535.092137] gpio-keys gpio_keys: failed to shutdown                          
[  535.283371] Unable to handle kernel NULL pointer dereference at virtual addre
ss 0000000000000638                                                             
[  535.292312] Mem abort info:                                                  
[  535.295145]   ESR = 0x0000000096000005                                       
[  535.298943]   EC = 0x25: DABT (current EL), IL = 32 bits                     
[  535.304334]   SET = 0, FnV = 0                                               
[  535.307431]   EA = 0, S1PTW = 0                                              
[  535.310610]   FSC = 0x05: level 1 translation fault                          
[  535.315558] Data abort info:                                                 
[  535.318474]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000                   
[  535.324039]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0                        
[  535.329164]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0                     
[  535.334562] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000041eae000        
[  535.341116] [0000000000000638] pgd=0000000000000000, p4d=0000000000000000, pu
d=0000000000000000                                                              
[  535.349960] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP          
[  535.356318] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_devi
ce rfcomm spidev cmac spi_bcm2835 algif_hash aes_arm64 aes_generic algif_skciphe
r af_alg bnep joydev als_ltr30x(O) gpio_pca953x industrialio_triggered_buffer kf
ifo_buf industrialio mipi_dsi(O) pinctrl_mcp23s08_i2c lis331dlh_i2c(O) rtc_pcf85
63 pinctrl_mcp23s08 brcmfmac_wcc hci_uart btbcm bluetooth bcm2835_codec(C) bcm28
35_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc vc_sm_cma(C) b
rcmfmac binfmt_misc rpi_hevc_dec v4l2_mem2mem videobuf2_dma_contig videobuf2_mem
ops videobuf2_v4l2 ecdh_generic brcmutil videodev ecc cfg80211 libaes snd_soc_si
mple_card snd_soc_simple_card_utils videobuf2_common rfkill raspberrypi_hwmon sn
d_bcm2835(C) mc i2c_bcm2835 i2c_brcmstb raspberrypi_gpiomem snd_soc_bcm2835_i2s 
gpio_keys nvmem_rmem snd_soc_wm8960 regmap_i2c i2c_dev ledtrig_pattern fuse dm_m
od ip_tables x_tables ipv6 v3d drm_shmem_helper vc4 gpu_sched snd_soc_hdmi_codec
 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd              
[  535.356510]  drm_display_helper drm_dma_helper cec drm_kms_helper drm drm_pan
el_orientation_quirks backlight uio_pdrv_genirq uio                             
[  535.458898] CPU: 2 UID: 0 PID: 1 Comm: systemd-shutdow Tainted: G        WC O
       6.12.20+rpt-rpi-v8 #1  Debian 1:6.12.20-1+rpt1~bpo12+1                   
[  535.471519] Tainted: [W]=WARN, [C]=CRAP, [O]=OOT_MODULE                      
[  535.476814] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)        
[  535.483345] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)  
[  535.490406] pc : vc4_hvs_unbind+0x20/0x160 [vc4]                             
[  535.495128] lr : component_unbind+0x40/0x70                                  
[  535.499374] sp : ffffffc08004ba70                                            
[  535.502730] x29: ffffffc08004ba70 x28: ffffff8040548000 x27: 0000000000000000
[  535.509974] x26: ffffffd39af5af38 x25: ffffffd39aef0bb8 x24: ffffffd39b58d9f0
[  535.517216] x23: ffffff808085e410 x22: ffffff8043c5e000 x21: ffffffffffffffff
[  535.524457] x20: 0000000000000000 x19: ffffff8081ec87c0 x18: ffffffc08163bc78
[  535.531699] x17: ffffffd39b6e7940 x16: ffffffd39a6bb048 x15: 0000000060922a08
[  535.538941] x14: 0000000000000004 x13: ffffffd39b5a7058 x12: 0000000000000000
[  535.546182] x11: ffffff8041b03298 x10: 0000000000001a40 x9 : ffffffd39a6b92a0
[  535.553424] x8 : ffffffc08004b990 x7 : 0000000000000000 x6 : ffffff808445fac8
[  535.560665] x5 : 0000000000150010 x4 : fffffffec209f2a0 x3 : ffffffd366bbc9a0
[  535.567906] x2 : ffffff8043c5e000 x1 : ffffff808085e410 x0 : ffffff8080853c10
[  535.575148] Call trace:                                                      
[  535.577622]  vc4_hvs_unbind+0x20/0x160 [vc4]                                 
[  535.581974]  component_unbind+0x40/0x70                                      
[  535.585864]  component_unbind_all+0xd0/0xe8                                  
[  535.590106]  vc4_component_unbind_all+0x20/0x38 [vc4]                        
[  535.595250]  devm_action_release+0x1c/0x30                                   
[  535.599406]  release_nodes+0x70/0x100                                        
[  535.603116]  devres_release_group+0xd4/0x150                                 
[  535.607444]  component_del+0xb8/0x170                                        
[  535.611156]  vc4_dsi_host_detach+0x30/0x58 [vc4]                             
[  535.615857]  mipi_dsi_detach+0x40/0x68                                       
[  535.619657]  i2c_md_shutdown+0x6c/0x98 [mipi_dsi]                            
[  535.624428]  i2c_device_shutdown+0x48/0x78                                   
[  535.628583]  device_shutdown+0x158/0x268                                     
[  535.632559]  kernel_power_off+0x40/0x88                                      
[  535.636447]  __do_sys_reboot+0x200/0x248                                     
[  535.640421]  __arm64_sys_reboot+0x2c/0x40                                    
[  535.644484]  invoke_syscall+0x50/0x120                                       
[  535.648287]  el0_svc_common.constprop.0+0x48/0xf0                            
[  535.653058]  do_el0_svc+0x24/0x38                                            
[  535.656418]  el0_svc+0x30/0xd0                                               
[  535.659515]  el0t_64_sync_handler+0x100/0x130                                
[  535.663934]  el0t_64_sync+0x190/0x198                                        
[  535.667647] Code: 910003fd a90153f3 f90013f5 f9403c34 (f9431e95)             
[  535.673828] ---[ end trace 0000000000000000 ]---                             
[  535.678538] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00
00000b                                                                          
[  535.686313] Kernel Offset: 0x1319e00000 from 0xffffffc080000000              
[  535.692314] PHYS_OFFSET: 0x0                                                 
[  535.695229] CPU features: 0x08,00002013,c0200000,0200421b                    
[  535.700703] Memory Limit: none                                               
~~~

I have tried this both with and without the E10-1 expansion attached -- it makes no difference.

The relevant `config.txt` lines are:

~~~
dtoverlay=dwc2,dr_mode=host
dtparam=ant2
disable_splash=1
ignore_lcd=1
dtoverlay=vc4-kms-v3d-pi4,noaudio
dtoverlay=i2c3,pins_4_5
gpio=13=pu
dtoverlay=reTerminal,tp_rotate=1
dtoverlay=seeed-2mic-voicecard

# fan on
gpio=23=op,pn,dh
~~~

If I comment out `dtoverlay=reTerminal,tp_rotate=1` and then reboot (after the obligatory power cycle because of the panic), then subsequent reboots and shutdowns work as expected.

Other potentially relevant information:

~~~
$ uname -a
Linux seeed 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

$ dkms status
bq24179_charger/0.1, 6.1.21-v8+, aarch64: installed
bq24179_charger/0.1, 6.12.20+rpt-rpi-v8, aarch64: installed
lis3lv02d/0.1, 6.1.21-v8+, aarch64: installed
lis3lv02d/0.1, 6.12.20+rpt-rpi-v8, aarch64: installed
ltr30x/0.1, 6.1.21-v8+, aarch64: installed
ltr30x/0.1, 6.12.20+rpt-rpi-v8, aarch64: installed
mipi_dsi/0.1, 6.1.21-v8+, aarch64: installed
mipi_dsi/0.1, 6.12.20+rpt-rpi-v8, aarch64: installed
seeed-voicecard/0.3: added

$ rpi-eeprom-config 
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
#BOOT_ORDER=0xf25641
BOOT_ORDER=0xf145

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
~~~

I have also updated the eeprom  using `rpiboot` (rpiboot/stable,now 20250401~081331 arm64) on a Pi 5 that is also fully up to date with the recovery image found in `/usr/share/rpiboot/recovery`.

Is there any additional information I can provide that might help narrow this down?

Thanks!

is-qian, 2025-04-10:
Thanks for your feedback, yes we are aware of this issue and are investigating it. You can see a discussion about it #124 .

asmagill, 2025-04-10:
I'll follow there than, thanks!

closed 2025-04-10 by asmagill