Seeed-Studio/seeed-linux-dtoverlays #129: Proper professional versioning

Kasperror, 2025-07-05:
As for every time I setup a new reTerminal DM device each time your software changes with breaking changes (setup scripts stop to work, different hardware/firmware issues every time) I ask you to add an automated versioning (tags) or frequent releases with bug fixes. I've already spoke about it in #122 with a proposal of using releases but with all due respect there is 3 years span between releases. Release 1.9 does not work well for reTerminal DM and 2.0 is already too new and breaks my automated scripts. Could you guys please provide a standard daily or on-main-push release?

Lesords, 2025-07-08:
I am very sorry for the inconvenience caused to you

Due to personnel changes and other reasons, the firmware may not be updated in time

We will actively maintain the firmware version in the future

Kasperror, 2025-07-08:
> I am very sorry for the inconvenience caused to you
> 
> Due to personnel changes and other reasons, the firmware may not be updated in time
> 
> We will actively maintain the firmware version in the future

It's not that the firmware is broken - we just can't keep our automation scripts running current dtovelays setup. The changes you bring to the script break automation so it would be much easier if we could just use pinned semantic versions. We don't want to pull the repo and then checkout some random commit. Would be much better if commits would finally pin a x.x.x version

Lesords, 2025-07-08:
If I understand correctly, it seems that you only need a specific version tag to pull the repository? No need for a release package?

closed 2025-07-30 by github-actions[bot]